MiCode/Xiaomi_Kernel_OpenSource #4699: Release source for sunstone please

vitivattXD, 2023-06-24:
Dear Xiaomi Team,

 I am a user of the Redmi Note 12 5G global Sunstone device and I am writing to request the release of the kernel and device trees for this device.

 As an enthusiast and developer, I am interested in customizing my device and contributing to the development of custom ROMs and kernels. However, I am unable to do so without access to the necessary kernel and device tree sources.

 I kindly request that Xiaomi release the kernel and device trees for the Redmi Note 12 5G Sunstone, as this will enable the development community to create custom ROMs and kernels for the device, and also improve the overall user experience.

 Thank you for your attention to this matter. I appreciate your continued support for the development community and look forward to hearing back from you.

Maheshwar, 2023-06-25:
+1 Xiaomi please provide kernel and device trees

Jay180-cyber, 2023-06-25:
> Dear Xiaomi Team,
>
> I am a user of the Redmi Note 12 5G global Sunstone device and I am writing to request the release of the kernel and device trees for this device.
>
> As an enthusiast and developer, [...]

+1

Jay180-cyber, 2023-06-25:
Kindly release source of sunstone

woooweee, 2023-06-25:
+1
:D