MiCode/Xiaomi_Kernel_OpenSource #4730: [Request] Poco X5 5G (Moonstone)

fikrie22, 2023-06-25:
Poco X5 5G Device trees and Kernel sources
Moonstone

mahe81868, 2023-06-27:
Dear Xiaomi Team,

I am writing to request the release of device trees for Poco X5 5G Moonstone. As a user of this device, I am interested in customizing and optimizing its performance, but I am unable to do so without access to the necessary resources.

I understand that Xiaomi has a strong commitment to open-source software, and I believe that releasing the device trees for Poco X5 5G Moonstone would be in line with this philosophy. By providing the necessary resources, you would enable the development of custom ROMs, kernels, and other modifications that would enhance the user experience for Poco X5 5G Moonstone owners.

I would like to emphasize that this request is not only for my personal benefit, but also for the benefit of the wider community of Poco X5 5G Moonstone users. By releasing the device trees, you would foster a more vibrant and active developer community, which would ultimately benefit Xiaomi as well.

I appreciate your time and consideration in this matter. I look forward to hearing from you soon regarding the release of device trees for Poco X5 5G Moonstone.

Thank you,

mahe81868, 2023-06-27:
https://t.me/Pocox5_moonstone