MiCode/Xiaomi_Kernel_OpenSource #4782: Xiaomi 12S Ultra kernel source

Nyanny, 2023-06-28:
Please release a kernel for Xiaomi 12S Ultra.
You have already [released a kernel for Xiaomi 12s](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/2614), which has an analogue of Qualcomm Snapdragon 8+ Gen1.

My topic is not the first, it was asked [here](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/4079) and [there](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/2964).

AlwaysZmx, 2023-09-09:
> Please release a kernel for Xiaomi 12S Ultra.
> You have already [released a kernel for Xiaomi 12s](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/2614), which has an analogue of Qualcomm [...]

+1

gituser765, 2023-09-29:
+1 please release sources for 12S Ultra (thor), much time has passed

AlwaysZmx, 2024-03-29:
+10086

RwOnke, 2025-05-27:
Is it ever going to be released?? 13's kernel already came out

RwOnke, 2025-08-02:
Still waiting, almost 2 yrs now

cornshiba, 2025-08-04:
> Still waiting, almost 2 yrs now

3 years have passed already, I am starting to doubt it

RwOnke, 2025-11-10:
Still waiting