MiCode/Xiaomi_Kernel_OpenSource #5340: Subject: Petition to Release FIXED/STABLE Source with Enabled FEAS for Redmi Note 11T Pro Device

ExWhyZed9, 2023-07-13:
Subject: Petition to Release FIXED/STABLE Source with Enabled FEAS for Redmi Note 11T Pro Device

Dear Xiaomi Developer Team,

I hope this letter finds you in good health and high spirits. I am writing to you as a loyal customer and an enthusiastic member of the Xiaomi community. I want to express my deep appreciation for the remarkable devices and exceptional software experiences that Xiaomi has consistently provided.

I am writing this petition on behalf of numerous Xiaomi Redmi Note 11T Pro device owners, who, like me, have been eagerly awaiting the release of a FIXED/STABLE source with enabled FEAS (Fast and Efficient Adaptive Security) for our devices. We believe that by addressing this concern, Xiaomi can significantly enhance the user experience and reaffirm its commitment to customer satisfaction.

We acknowledge the efforts made by the Xiaomi developer team in creating a robust and feature-rich software ecosystem. However, many Redmi Note 11T Pro device owners have encountered persistent issues related to stability, performance, and security. These issues have hampered our ability to fully enjoy the potential of our devices and have left us feeling frustrated.

We firmly believe that enabling FEAS on the Redmi Note 11T Pro device would greatly alleviate these concerns. FEAS has been widely recognized as a powerful security feature that not only ensures the safety of user data but also optimizes device performance. By implementing FEAS on the Redmi Note 11T Pro device, Xiaomi can demonstrate its commitment to user privacy, data protection, and overall system stability.

We understand that developing and fine-tuning software for a vast range of devices can be a challenging task. However, we kindly request that Xiaomi allocate the necessary resources and prioritize the development of a FIXED/STABLE source for the Redmi Note 11T Pro. We firmly believe that this initiative would not only benefit Redmi Note 11T Pro device owners but also strengthen Xiaomi's reputation as a customer-centric and innovative company.

We also asked for this three times, with no success, because of a broken kernel release, and broken kernel with FEAS deleted. Many people sell this device after purchased because this issue makes very difficult the Custom ROM installation process, and the community developers aren't able to develop a free-bug ROM/tweak (also there are no testers). 

In conclusion, we, the undersigned Redmi Note 11T Pro device owners, respectfully request Xiaomi to release a FIXED/STABLE source with enabled FEAS for our devices. We believe that this step would significantly enhance the overall user experience, address existing concerns, and solidify Xiaomi's position as a leading provider of cutting-edge technology.

Thank you for your time and consideration. We look forward to your positive response and eagerly anticipate the release of the requested software update.

Sincerely,
ExWhyZed9

AlexFisher7918g, 2023-07-14:
Yeah,we need it, please

plnxin, 2023-07-14:
Yeah,we need it, please

ditstyle-yang, 2023-07-14:
I hope so

JaneValleyhasBtaken, 2023-07-14:
Yes this is very much needed

Ekyfig, 2023-07-15:
We need it.I support you.

AlexFisher7918g, 2023-07-15:
Come on,guys, let's speak for ourselves!