MiCode/Xiaomi_Kernel_OpenSource #5382: dagu-s-oss , Can't compile successfully

jxwdsb, 2023-07-29:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/dagu-s-oss

Can't compile successfully, can you give me a tutorial? According to the wiki also can not succeed

WARNING: EXPORT symbol "gsi_write_channel_scratch" [vmlinux] version generation failed, symbol will not be versioned.
ld.lld: error: relocation R_AARCH64_ABS32 cannot be used against symbol __crc_gsi_write_channel_scratch; recompile with -fPIC
>>> defined in built-in.a(drivers/platform/msm/gsi/gsi.o)
>>> referenced by gsi.c
>>>               drivers/platform/msm/gsi/gsi.o:(___kcrctab+gsi_write_channel_scratch+0x0) in archive built-in.a
/home/linux/dagu-s-oss/Makefile:1196: recipe for target 'vmlinux' failed
make[1]: *** [vmlinux] Error 1

shidalao, 2024-02-07:
solution:
`sed` -i 's/static union __packed gsi_channel_scratch/static union gsi_channel_scratch/' drivers/platform/msm/gsi/gsi.c

let me know if it helps or not

shidalao, 2024-03-13:
> > solution: `sed` -i 's/static union __packed gsi_channel_scratch/static union gsi_channel_scratch/' drivers/platform/msm/gsi/gsi.c
> > let me know if it helps or not
> 
> it will compile successfully . But cannot boot (not bootloop,just stuck in the MI logo)
> 
> Branch pipa-t-oss , same problem

Then it should be another problem, such as module signature...

dirt2022, 2024-03-16:
But how can I solve the boot stuck problem?