MiCode/Xiaomi_Kernel_OpenSource #5387: Please release RedmiNote11 ( spes/spesn) Android T kernel

luvcamellia, 2023-07-30:
Android 11 kernel of spes is very old and missing driver of camera and flashlight

Guilher89, 2023-09-30:
I hope this message finds you well. I'm contacting you for information and guidance on compiling the SPES kernel. I am interested in understanding the compilation process and its specificities.

So that I can gain a more comprehensive understanding, I would like to ask you a few specific questions:

Toolchain Used: Which toolchain (set of compilation tools) did you use to compile the SPES kernel? Was it a specific toolchain recommended by the manufacturer or one of your choice?

Compilation Script: During the compilation process, did you use any custom kernel compilation script or a manufacturer-supplied script? If yes, could you share details about this script or its settings?

Specific Settings: Were there any special settings or adjustments you needed to make when compiling the SPES kernel, especially related to specific hardware or optimizations?

Issues or Challenges: While compiling the kernel, did you face any specific issues or challenges that you would like to highlight? How did you resolve these issues, if any?

If you are willing to share your experiences and knowledge about compiling the SPES kernel, I would be extremely grateful. Your information will be invaluable to my own compiling journey.

Please know that any details or guidance you can provide will be highly appreciated. If you prefer to share this information in a more private setting, I am happy to exchange private messages or emails, whichever is most convenient for you.

Thank you in advance for your help and collaboration. Your knowledge will be of great benefit.

Guilher89, 2023-09-30:
copy the techpack from https://github.com/PixelExperience-Devices/kernel_xiaomi_sm6225.git and then build the kernel