MiCode/Xiaomi_Kernel_OpenSource #5911: Release::: Kernel sources for Redmi Note 12 5g (Sunstone)

KOAwesome, 2023-08-16:
Greetings, Xiaomi Team

I am writing as a Redmi Note 12 5G India Sunstone user to ask for the release of the kernel and device trees for this particular handset.

I'm interested in customizing my device and helping to create custom ROMs and kernels because I'm an enthusiast and developer. Without having access to the required kernel and device tree sources, I am unable to achieve it, though.

I ask Xiaomi to make the Redmi Note 12 5G Sunstone's kernel and device trees available, as doing so will allow the development community to make custom ROMs and kernels for the smartphone and enhance the user experience in general.

I appreciate you giving this issue your attention. I value your ongoing assistance with the information.

Pra7ul, 2023-10-09:
Please help community xiaomi

Last-Hub-vode, 2024-05-20:
The GPL requires them to release the source of the kernel or any other GPL'd components.