MiCode/Xiaomi_Kernel_OpenSource #5915: [APPEAL] Release FULL KERNEL SOURCE FOR MARBLE (POCO F5 / REDMI NOTE 12 TURBO)

spike0en, 2023-08-18:
Dear Xiaomi Officials,

We, the users and custom ROM community, express our concerns regarding the incomplete (half-baked) June release by @Jshengyin (marble-s-oss). We believe this release does not meet essential requirements for effective device development.

As open source enthusiasts, we believe in upholding GPL rules. We appreciate Xiaomi's commitment to open source development and request your support in releasing complete sources as per the GPL rules. Open source thrives on collaboration, accountability, and transparency. Complete sources empower developers, foster innovation, and enhance user experiences.

We understand the challenges in managing and releasing source codes, but comprehensive and accurate availability is crucial for developer and user satisfaction. Please prioritize the release of complete sources to enable remarkable custom ROMs.

By adhering to open source principles, Xiaomi can strengthen its reputation and user trust. We await a positive response and hope Xiaomi sets a benchmark for other manufacturers.

Thank you for your commitment to open source development. We eagerly anticipate your response and look forward to collaboration.

Sincerely,
[POCO F5 / REDMI NOTE 12 TURBO COMMUNITY](t.me/POCOF5_official)
[image]

2036648790, 2023-08-18:
请发布完整源代码

freeslam, 2023-08-18:
How can we hope for good customs roms and our Poco F5 to last longer if you don't give a us a proper kernel source and vendor ! I know you can do better !

Sabbir2k15, 2023-08-19:
Please Release ASAP

hibreck, 2023-08-19:
Please

imsahilansarii, 2023-08-19:
Please release

thanveeralisayed, 2023-08-19:
Yes. Please release the full kernel sources for poco f5 / Redmi note 12 turbo. (Marble)

keijiyuy, 2023-08-19:
Please release the full kernel sources for poco f5 (Marble), everybody wait for this.

luoxuxi, 2023-08-24:
full kernel sources plz

UnFunnyGuy, 2023-08-25:
Please Release Kernel Source ffs

Sudeepxl20, 2023-08-28:
Release the source

t01h01, 2023-08-28:
Release source plz

2907727365, 2023-08-29:
发放完整的源码

Mustafarexdo, 2023-08-29:
Please Release the full source

Mohitshakya81, 2023-09-03:
Please Release full kernel source

Chizkayk, 2023-09-12:
Full kernel source?🥺

freeslam, 2023-10-07:
Nothing yet ? Common guys we want full support for this device !

Satyam709, 2023-10-24:
Release full kernel source

closed 2024-12-31 by spike0en