MiCode/Xiaomi_Kernel_OpenSource #6606: Please Update Poco X5 Pro 5G (REDWOOD) Kernel Source

FastGecko5, 2023-08-26:
Currently available kernel is incompatible with recent revisions of the phone (batch numbers 2023.06 and 2023.07) due to a hardware revision that changed the display. 

Please release updated kernel sources so that development may continue for this phone.

jpm92, 2023-10-01:
Dear Xiaomi,

In your devices, you are distributing executable copies of the Linux kernel, which is licensed under the GNU General Public License version 2.0 (GPLv2). Under section 3 of this license, anyone who distributes executable form of the software must provide the users with the complete source code to the software on request.
I thus request that you fulfill the license's terms by providing the updated and complete source code of Poco X5 Pro (redwood). Please be advised that failure to comply with the terms of this license results in permanent termination of your rights to modify and distribute the Linux kernel under section 4 of the GPL.

harshvardhan-shah, 2023-10-21:
Dear Xiaomi & Poco,

You distribute executable copies of the Linux kernel in your devices, licensed under the GNU General Public License version 2.0 (GPLv2). Under section 3 of this license, anyone who distributes an executable form of the software must provide the users with the complete source code to the software on request.
I ask that you fulfill the license's terms by providing the updated and complete source code of Poco X5 Pro (redwood). Please keep in mind that failure to comply with this license's terms results in permanently ending your rights to modify and distribute the Linux kernel under section 4 of the GPL.

mansoor4455, 2024-03-26:
Dear Xiaomi,

In your devices, you are distributing executable copies of the Linux kernel, which is licensed under the GNU General Public License version 2.0 (GPLv2). Under section 3 of this license, anyone who distributes executable form of the software must provide the users with the complete source code to the software on request.
I thus request that you fulfill the license's terms by providing the updated and complete source code of Poco X5 Pro (redwood). Please be advised that failure to comply with the terms of this license results in permanent termination of your rights to modify and distribute the Linux kernel under section 4 of the GPL.

Ibadriansyah, 2025-02-25:
Dear Xiaomi & Poco 

Please update to GKI V2