MiCode/Xiaomi_Kernel_OpenSource #6714: Xiaomi 11i/Redmi note 11 pro PISSARRO kernel update

Moazam98, 2023-09-02:
Dear Xiaomi,
       On the behalf of Xiaomi 11i/Redmi note 11 pro(PISSARRO) Community, I am demanding for the kernel update of this device. The previous kernel which was released has Touch and Fast charging issues. To promote healthy development of custom ROMs for this device we need a working kernel for starters....
       We wont stop asking for the kernel unless and until it is released so better update it fast, its already been more than a year now. And you are violating GPL, which is not acceptable at all. Again a reminder for UPDATING THE KERNEL SOURCE FOR PISSARRO.
Yours sincerely,
Moazam Khan

binboupan, 2023-09-03:
Don't spam!! One issue is enough!

Moazam98, 2023-09-03:
> Don't spam!! One issue is enough!

If that would have been enough we would have got our solution or atleast one fo the issues(broken kernel or blue tint issue due to a software bug)