MiCode/Xiaomi_Kernel_OpenSource #6772: [REQUEST] FULL & COMPLETE KERNEL SOURCE CODE FOR REDMI NOTE 12 TURBO/POCO F5 (marble)

Nixsuki, 2023-10-03:
THIS IS TO REQUEST THE FULL AND COMPLETE KERNEL SOURCE CODE FOR REDMI NOTE 12 TURBO/POCO F5
THE DEVICE RELEASE WAY BACK MAY 2023 AND NOW ALMOST 5 MONTHS NO COMPLETE MODULE AND SOURCE CODES! THIS IS A SERIOUS GPL VIOLATION FOR UNRESPONSIVE AND INCOMPETENCY!
YOU ARE BRINGING PEOPLE AND COSUMERS DISAPPOINTMENTS AND ELSE MAYBE SOMEDAY PEOPLE IN OPEN SOURCE COMMUNITY BOYCOTT YOUR (F) SERIES FOR UNPROFESSIONAL LATE RELASING OF THE COMPLETE SOURCE!

freeslam, 2023-10-11:
Common guys it's the mauin reason why i buy xiaomi phone ! Give us some love with a full & complet kernel source so we can enoy our new device to the fullest and be sure to buy from you again !

Stephanizgo, 2023-10-17:
I have seen that the marble-s-oss kernel works, people made roms with it

freeslam, 2023-10-17:
> I have seen that the marble-s-oss kernel works, people made roms with it

I know but it's incomplete ! We want the full kernel to have good roms.