MiCode/Xiaomi_Kernel_OpenSource #6785: Problems when compiling the titing kernel

500zxy, 2023-10-15:
I tried to compile the kernel source code of the Android device code named titing. The compilation configuration file I used is the gki_defconfig in the configs directory. The error is as follows:

Drivers/misc/Kconfig: 540: can't open file "drivers/misc/hwid/Kconfig"
Make [2]: *** [../scripts/kconfig/Makefile: 88: gki_defconfig] error 1
Make [1 ]: *** [/ home/xiaohua/desktop/Kernel/Makefile: 619: gki_defconfig] error 2
Make [1]: leave directory "/home/xiaohua/desktop/Kernel/output"
Make : *** [ Makefile: 185: __sub-make] error 2

500zxy, 2023-10-15:
请开源完整的内核源码

KisaragiAkira, 2023-10-24:
I have the same problem.
When I build fuxi kernel, it notify me missing driver/mics/hwid and missing driver/staging/binder_pro. 
Remove relation Makefile and Kconfig will can be built it, but I think the kenel source code is not complete.

I do not know why Xiaomi not opens ouce hwid, but you can try to remove hwid from driver/misc/Kconfig and driver/misc/Makefile.

yomihime, 2023-10-24:
> I have the same problem. When I build fuxi kernel, it notify me missing driver/mics/hwid and missing driver/staging/binder_pro. Remove relation Makefile and Kconfig will can be built it, but I think [...]

Pin it, it's from my deleted account. (an unuse account)

acama, 2024-12-05:
> Pin it, it's from my deleted account. (an unuse account)

Can you specify the steps you used to build it ? seems I can't even build with those lines removed or I am missing something. Which toolchain you used and commands ?

yomihime, 2024-12-06:
> Can you specify the steps you used to build it ? seems I can't even build with those lines removed or I am missing something. Which toolchain you used and commands ?

Sorry, I use internal tool to build it. You can try to combin with Qualcomm AOSP. May it work.