MiCode/Xiaomi_Kernel_OpenSource #6829: [Request] Release kernel source code for Redmi Note 12 5G (sunstone)

Shreedhan003, 2023-11-15:
Greetings, Xiaomi Team

I am writing as a Redmi Note 12 5G (sunstone) user to ask for the release of the kernel and device trees for this particular handset.

I'm interested in customizing my device and helping to create custom ROMs and kernels because I'm an enthusiast and developer. Without having access to the required kernel and device tree sources, I am unable to achieve it, though.

I ask Xiaomi to make the Redmi Note 12 5G (sunstone) kernel and device trees available, as doing so will allow the development community to make custom ROMs and kernels for the smartphone and enhance the user experience in general.

I appreciate you giving this issue your attention. I value your ongoing assistance with the information.

luffy-2510, 2023-11-15:
Hello sir can you give me your tg username

Sentifool, 2023-12-13:
Greetings, Xiaomi Team

I am writing as a Redmi Note 12 5G (sunstone) user to ask for the release of the kernel and device trees for this particular handset.

I'm interested in customizing my device and helping to create custom ROMs and kernels because I'm an enthusiast and developer. Without having access to the required kernel and device tree sources, I am unable to achieve it, though.

I ask Xiaomi to make the Redmi Note 12 5G (sunstone) kernel and device trees available, as doing so will allow the development community to make custom ROMs and kernels for the smartphone and enhance the user experience in general.

I appreciate you giving this issue your attention. I value your ongoing assistance with the information.

Shreedhan003, 2024-04-27:
Maybe not today, not tomorrow or not even day after that but trust me @mi-code there will be a day when you will get hit hard because of this GPL related negligence and trust me I'll be waiting for that downfall one day

closed 2024-04-27 by Shreedhan003