MiCode/Xiaomi_Kernel_OpenSource #6932: Kernel flashing on ishtar - A/B partitioned devices with dynamic partitions

lupo-ch, 2024-02-17:
Hello, Kernel Developers!

I hope that somebody can help. :) 
I'm trying to flash the kernel I build for ishtar (Xiaomi 13 Ultra).
I have fixed few errors which I have got with the Xiaomi github released source code and compiled the kernel OK.

I have ready: Image.gz, dtb.img and dtbo.img

For all my old phones I use AnyKernel3 latest version form github to flash my custom kernels, but here with ishtar I have some very strange error.
When flashing the kernel via TWRP I receive this error:

New image larger than target partition. Aborting...

The error is from the AnyKernel3 in tools/ak3-core.sh

My kernel image is not so big actually
-rw-r--r-- 1 root root 14113137 фев 13 17:26 Image.gz
and the boot.img is much much bigger 

I'm guessing it's some other issue here ...

The ishtar device is using dynamic partitions and A/B slots.
I have already activated the:
is_slot_device=1; in anykernel.sh 

I don't have the ramdisk available in order to build a boot.img and actually haven't done that beforE.
I have flashed all my kernels with  AnyKernel3

Has anyone here dealt with similar issues on A/B partitioned devices or dynamic partitions, particularly with ishtar or similar devices?
Any pointers or suggestions for troubleshooting this would be greatly appreciated.
Thank you in advance!

Wvwvwvwvwv, 2024-06-18:
Hello, I’m really looking for a custom kernel for mi 13 ultra, if you managed to create it, and I tried but I can’t, could you be so kind as to share with me the kernel and the source code for it, I will be eternally grateful to you.

Wvwvwvwvwv, 2024-06-18:
If you need a tester, I will be very happy to join you