MiCode/Xiaomi_Kernel_OpenSource #8268: FIX ALREADY!! fix camellia-r-oss and release camellia-t-oss | 修复camellia-r-oss并发布camellia-t-oss!!!

gartriss, 2024-04-03:


1593358539, 2024-04-04:
# Please stop.
Don't spam this repository with these meaningless issues. Xiaomi is definitely going to ignore more legitimate requests if you continue spamming. They're a big company and have no responsibility to open source any of their code.

You have to acknowledge the fact that you're essentially asking a large corporation to **_publish trade secrets._**

Once again, please stop. Do not spam these repositories. This isn't even an "issue" in the first place. Write to Xiaomi directly, and again, do not spam.

gartriss, 2024-04-04:
> # Please stop.
>
> Don't spam this repository with these meaningless issues. Xiaomi is definitely going to ignore more legitimate requests if you continue spamming. They're a big company and have no [...]

Please stop. 
I need stupid reason to write to github
Once again please stop :D

ayushqui, 2024-04-12:
Wow

matu6968, 2024-05-18:
> You have to acknowledge the fact that you're essentially asking a large corporation to **_publish trade secrets._**
> 
You do realize that the original Linux kernel source uses GPLv2 license meaning they have to release it either ways as they are violating the GPL by not providing compilable code or even functional code. FSF (Free Software Foundation) can take action against it as they do care about open source licenses.
So the best way to deal with it is by telling FSF that someone violated GPL terms and to force them to fix it.

1593358539, 2024-05-19:
This is an oversimplified and misinformative view on GPLv2's terms. The GPLv2 license has *several* loopholes which allow the proprietary redistribution of modified OSS source code. It is absolutely a trade secret. **Leaking and reverse-engineering it without permission from the proprietor, is illegal.** I highly doubt Xiaomi, a huge tech giant, would overlook this.

Take note of Red Hat's closure of RHEL (and other product...) source code which contain source available forks and moficiations to not just the Linux kernel, but other GPLv2 licensed projects. 

Don't like it? Don't use the software.

matu6968, 2024-11-19:
> This is an oversimplified and misinformative view on GPLv2's terms.

GitHub uses license summary from [choosealicense.com](https://choosealicense.com/licenses/gpl-2.0/) it and is that a problem?

> The GPLv2 license has _several_ loopholes which allow the proprietary redistribution of modified OSS source code. It is absolutely a trade secret. 

Loopholes exist if the modified OSS source code is licensed under a license which inhibites that the source **must be released.** Just think at hundreds of projects that use modified GPL open source code and they don't release the modified code, and that one day they can get sued over it, hence why honest manufacturers like Motorola and Samsung release the kernel code to prevent legal trouble from their side

> **Leaking and reverse-engineering it without permission from the proprietor, is illegal.** 

Did you also not think of  loopholes existing in [both in the US](https://en.wikipedia.org/wiki/Reverse_engineering#United_States) and [EU](https://en.wikipedia.org/wiki/Reverse_engineering#European_Union)??? + someone [already reverse engineered the code to work on kernel 4.19](https://vt.tiktok.com/ZNe3RLNCL) yet they don't themselves release the code to it so it is a cat and mouse game between them and possibly the FSF.