MiCode/Xiaomi_Kernel_OpenSource #27775: Release kernel source of Poco c51/Redmi a2/Redmi a2+  codenae (water)

RG-98, 2024-08-06:
Dear Xiaomi Team,

I am a user of the Poco c51/Redmi a2/Redmi a2+ india water device and I am writing to request the release of the kernel and device trees for this device.

As an enthusiast and developer, I am interested in customizing my device and contributing to the development of custom ROMs and kernels. However, I am unable to do so without access to the necessary kernel and device tree sources.

I kindly request that Xiaomi release the kernel and device trees for the Poco c51/Redmi a2/Redmi a2+ water, as this will enable the development community to create custom ROMs and kernels for the device, and also improve the overall user experience.

Thank you for your attention to this matter. I appreciate your continued support for the development community and look forward to hearing back from you.

AndrewUESP31, 2024-12-31:
Hi, I'm trying to make a custom TWRP recovery for a Redmi A2 Plus. I haven't found the kernel tree anywhere. Maybe you can analyze the boot.img file of the official ROM of the phone. With that we can make a BoardConfig.mk and made custom ROMs for the phones.

RG-98, 2024-12-31:
Bro we already completed custom recovery of poco c51 we need a server or
high end pc to make custom rom can you help ?

IslamKouadria, 2025-04-04:
hello any budy make the recovery if you did please send me the file