MiCode/Xiaomi_Kernel_OpenSource #27784: KERNEL SOURCE FOR REDMI-A2/A3/POCO-C51

cranckv3, 2024-08-13:
I am a user of the Redmi A2 4G india Water device and I am writing to request the release of the kernel and device trees for this device.

As an enthusiast and developer, I am interested in customizing my device and contributing to the development of custom ROMs and kernels. However, I am unable to do so without access to the necessary kernel and device tree sources.

I kindly request that Xiaomi release the kernel and device trees for the Redmi A2 4G Water, as this will enable the development community to create custom ROMs and kernels for the device, and also improve the overall user experience.

Thank you for your attention to this matter. I appreciate your continued support for the development community and look forward to hearing back from you.

AsTechpro20, 2024-08-17:
I completely agree with this request

Dev-Ayush-Gupta, 2024-10-13:
Ya, we need it

PermisiveAuditor, 2026-04-01:
Wow, this is really interesting! It's the same device, which I also want to learn how to modify, although I don't know where to start. Can I help you with anything? Are you still researching how to proceed? Best regards.