MiCode/Xiaomi_Kernel_OpenSource #27845: isn't it against GPL or whatever for not releasing the source code?

winvistalover, 2024-09-28:


ayushqui, 2024-09-28:
Yes indeed

winbo-yml-exe, 2024-09-28:
Yeah, we can actually go to https://gpl-violations.org and report them.

ayushqui, 2024-09-28:
How to report there but .

ayushqui, 2024-09-28:
Here mail here :- info@gpl-violations.org

winbo-yml-exe, 2024-10-02:
[image]

winbo-yml-exe, 2024-10-02:
If she doesn't do it, I will be emailing https://gpl-violations.org

winbo-yml-exe, 2024-10-02:
> Yes go for it

She told me to submit feedback asking Xiaomi to release it, and also said most devices got their kernel source released because of it, I will wait until that feedback gets reviewed.

ayushvps, 2024-10-02:
Can u provide ss

winvistalover, 2024-10-09:
hi, anyone alive?

hilmidp, 2024-10-10:
so how did it go guys?

winbo-yml-exe, 2024-10-26:
> so how did it go guys?

still no progress on getting source code, prebuilt is out tho