MiCode/Xiaomi_Kernel_OpenSource #28626: Release::: Kernel sources for Redmi 12 5g (Sky/River)

schoolgadou, 2024-10-27:
Greetings to thou Xiaomi Team.

I'm bringing up the fact that we don't have SKY device trees and kernels.

As a developer and enthusiast, I'm interested in personalising my device and contributing to the development of unique ROMs and kernels. However, I cannot accomplish it without access to the necessary kernel and device tree files.

In order to improve the overall user experience and enable the development community to create custom ROMs and kernels for the smartphone, I request that Xiaomi make the Redmi 12 5G SKY's kernel and device trees public.

Thank you for taking the time to consider this matter. I appreciate your continued help with the data.

winbo-yml-exe, 2024-10-27:
> Greetings to thou Xiaomi Team.
>
> I'm bringing up the fact that we don't have SKY device trees and kernels.
>
> As a developer and enthusiast, I'm interested in personalising my device and contributing [...]

The Device Tree is their code, they don't have to give it.

schoolgadou, 2024-10-27:
> they don't have to give 
Some dev friend of mine said they wish they had it. I just want the kernel, are you saying devs don't need it or are you saying its their private code.