MiCode/Xiaomi_Kernel_OpenSource #36474: [GNU Kernel Source Code request] Redmi 13C 4G (gale/gust)

MIKAILREI, 2024-10-29:
Hello, It's been almost a year since the Redmi 13C 4G got released, and we, the community, need the source code of the kernel. There's a lot of issues on here asking for the source code, and the Linux kernel is GNU, so please release the source code of the kernel.

schoolgadou, 2024-10-30:
holy shit, stop spamming

Vaibhavbhor404, 2024-10-30:
> holy shit, stop spamming

Bro come on , we are not spamming just we are requesting to get kernel source as per GPL rule.

spellydev, 2024-10-30:
> holy shit, stop spamming

we are doing this for a good cause

Vaibhavbhor404, 2024-10-30:
> > holy shit, stop spamming
> 
> we are doing this for a good cause

Yeah also Xiaomi didn't follow GPL rule

Efeisot, 2024-10-31:
> > holy shit, stop spamming
> 
> Bro come on , we are not spamming just we are requesting to get kernel source as per GPL rule.

> > holy shit, stop spamming
> 
> Bro come on , we are not spamming just we are requesting to get kernel source as per GPL rule.

Me and my community also have a poco pad and Xiaomi didn't released kernel source for it nearly 7 months. BUT THIS IS NOT X, YOU F*CKING FR*AKS, YOU HAVE OPENED MORE THAN 8000 ISSUE PAGES IN 2-3 DAYS FOR YOUR ONE DEVICE! DO YOU THINK OVER 8000 ISSUE PAGES ARE NOT SPAM ENOUGH? YOU MAY BE SEEKING YOUR RIGHTS DUE TO GPL LICENSE (WE ARE TOO); BUT CANT YOU ACT LIKE A HUMAN AND RESPECT THE ISSUE PAGES OPENED FOR OTHER DEVICES! WHAT DO YOU HAVE TO LOSE? THERE IS NO MEANING TO YOUR STUPID SPAM IF XIAOMI WON'T LOOK AT IT ANYWAY, AND IF IT DOES, YOU WILL BE PUTTING THE DEVELOPERS OF OTHER DEVICES IN THE BACKGROUND FOR NOTHING!

Vaibhavbhor404, 2024-10-31:
> Me and my community also have a poco pad and Xiaomi didn't released kernel source for it nearly 7 months. BUT THIS IS NOT X, YOU F_CKING FR_AKS, YOU HAVE OPENED MORE THAN 8000 ISSUE PAGES IN 2-3 [...]

What the hell who said you that I spam 8000 issue I only raise mine one so stop doing nonsense

Efeisot, 2024-10-31:
> What the hell who said you that I spam 8000 issue I only raise mine one so stop doing nonsense

If you (@Vaibhavbhor404) didn't understand, I use the term "you" not only for you, but for all three of you (@Vaibhavbhor404, @spellydev, @MIKAILREI), including that idiot who is spamming the issue topics right now (even you [@Vaibhavbhor404] use the term "we", it's stupid to talk about it here). Besides, i am currently in the issue topic of the idiot (@MIKAILREI) who made the spam, not in your (@Vaibhavbhor404's) issue topic (even if you [@Vaibhavbhor404] opened only one issue, aren't the 8000 "we want the kernel source for Redmi 13c 4g" issues in the issue category not enough for you, so you opened an issue on top of that?)

IF YOU (@Vaibhavbhor404) ARE STILL TENDENCY TO BE STUBBORN, LET ME SUMMIT IT: WHEN THERE ARE 8000 ISSUE TOPICS, DON'T OPEN AN EXTRA ONE. IF YOU (@Vaibhavbhor404) DO, AND YOU (@Vaibhavbhor404) WILL SUPPORT THOSE FOOLISH TOPICS WHO OPEN THESE SPAM TOPICS, THEN DO NOT TRY TO EXCUSE YOURSELF (@Vaibhavbhor404).

enis26, 2024-10-31:
> > holy shit, stop spamming
> 
> Bro come on , we are not spamming just we are requesting to get kernel source as per GPL rule.

That wont change anything you fucking dumbass

Vaibhavbhor404, 2024-10-31:
> > > holy shit, stop spamming
> > 
> > 
> > Bro come on , we are not spamming just we are requesting to get kernel source as per GPL rule.
> 
> That wont change anything you fucking dumbass

Mind your language @enis26 I'm just asking not abusing so get your sucking tongue in your control else I also know another ways to handle the bs guys like you

spellydev, 2024-10-31:
y

> If you (@Vaibhavbhor404) didn't understand, I use the term "you" not only for you, but for all three of you (@Vaibhavbhor404, @spellydev, @MIKAILREI), including that idiot who is spamming the issue [...]

you all could do this too... who is stopping you lol

Vaibhavbhor404, 2024-10-31:
> IF YOU (@Vaibhavbhor404) ARE STILL TENDENCY TO BE STUBBORN, LET ME SUMMIT IT: WHEN THERE ARE 8000 ISSUE TOPICS, DON'T OPEN AN EXTRA ONE. IF YOU (@Vaibhavbhor404) DO, AND YOU (@Vaibhavbhor404) WILL [...]

As I already said I'm not going to open any issue * I'm just giving my opinion*

MIKAILREI, 2024-11-01:
> If you (@Vaibhavbhor404) didn't understand, I use the term "you" not only for you, but for all three of you (@Vaibhavbhor404, @spellydev, @MIKAILREI), including that idiot who is spamming the issue [...]

TO BE FAIR, I DIDNT MAKE THE SPAM, @winbo_yaml make the script.

MIKAILREI, 2024-11-01:
also fuck u @Efeisot NO ONE CARES ABOUT U AND NEVER WILL.

closed 2024-11-01 by MIKAILREI

at3mic, 2026-09-28:
> If you ([@Vaibhavbhor404](https://github.com/Vaibhavbhor404)) didn't understand, I use the term "you" not only for you, but for all three of you [...]

Thanks for the shout out. ❤️ 

Regards,
https://github.com/ghost