MiCode/Xiaomi_Kernel_OpenSource #36480: [GNU KERNEL SOURCE DEMAND] Redmi 13C 5G / POCO M6 5G (air)

Pranavsucksatlife, 2024-11-17:
Hello, Xiaomi!

We are writing to you regarding an issue with the Xiaomi Redmi 13C 5G / POCO M6 5G smartphones (air). The kernel source for these devices is incomplete and broken and OUTDATED, as seen in the repository:
https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/air-t-oss

For instance, critical components such as several camera drivers, DRM, and touchscreen drivers are missing. This is a direct violation of the GNU GPL. The Linux kernel used in the Android project is protected under the GPL license, which requires Xiaomi to publish the complete source code and relevant branches for each Android device released.

Our attempts to fix the kernel have been unsuccessful due to these deficiencies, leaving you obligated to address and resolve the issue. A functional and complete kernel source would be an invaluable contribution to the community, particularly developers working on custom ROMs and kernel optimizations.

Unfortunately, your Global Support team has repeatedly ignored our requests and responded only with template answers. This situation is unacceptable, especially considering that these devices are not yet EOL (End of Life). You still have the opportunity—and responsibility—to rectify this.

We urge you to address this issue and release a complete, functional kernel source for the Xiaomi Redmi 13C 5G and POCO M6 5G devices immediately.

Best regards,
Pranav

winbo-yml-exe, 2024-11-17:
they didn't even release the ks for gale, and even if they did, it would probably be broken too

btw look at the issues i made here

Pranavsucksatlife, 2024-11-17:
Where?

> they didn't even release the ks for gale, and even if they did, it would probably be broken too
> 
> btw look at the issues i made here