MiCode/Xiaomi_Kernel_OpenSource #36486: How to Build?

Fighteros, 2024-11-23:
I have been trying to build the Kernel for Vayu from branch vayu-r-oss
on GitHub actions using Android-Kernel-Builder by @PhamtomK12
and i get these errors

```
make[1]: Entering directory '/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/out'
  GEN     ./Makefile
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  GEN     ./Makefile
  WRAP    arch/arm64/include/generated/uapi/asm/errno.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctl.h
  WRAP    arch/arm64/include/generated/uapi/asm/ioctls.h
  CHK     include/generated/uapi/linux/version.h
  WRAP    arch/arm64/include/generated/uapi/asm/ipcbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/kvm_para.h
  WRAP    arch/arm64/include/generated/uapi/asm/mman.h
  UPD     include/generated/uapi/linux/version.h
  WRAP    arch/arm64/include/generated/uapi/asm/msgbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/poll.h
  WRAP    arch/arm64/include/generated/uapi/asm/resource.h
  WRAP    arch/arm64/include/generated/uapi/asm/sembuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/shmbuf.h
  WRAP    arch/arm64/include/generated/uapi/asm/socket.h
  WRAP    arch/arm64/include/generated/uapi/asm/sockios.h
  WRAP    arch/arm64/include/generated/uapi/asm/swab.h
  WRAP    arch/arm64/include/generated/uapi/asm/termbits.h
  WRAP    arch/arm64/include/generated/uapi/asm/termios.h
  WRAP    arch/arm64/include/generated/uapi/asm/types.h
  HOSTCC  scripts/basic/fixdep
  WRAP    arch/arm64/include/generated/asm/bugs.h
  WRAP    arch/arm64/include/generated/asm/clkdev.h
  WRAP    arch/arm64/include/generated/asm/delay.h
  WRAP    arch/arm64/include/generated/asm/div64.h
  WRAP    arch/arm64/include/generated/asm/dma.h
  WRAP    arch/arm64/include/generated/asm/dma-contiguous.h
  WRAP    arch/arm64/include/generated/asm/early_ioremap.h
  WRAP    arch/arm64/include/generated/asm/emergency-restart.h
  WRAP    arch/arm64/include/generated/asm/hw_irq.h
  WRAP    arch/arm64/include/generated/asm/irq_regs.h
  WRAP    arch/arm64/include/generated/asm/kdebug.h
  WRAP    arch/arm64/include/generated/asm/kmap_types.h
  WRAP    arch/arm64/include/generated/asm/local.h
  WRAP    arch/arm64/include/generated/asm/local64.h
  WRAP    arch/arm64/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm64/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm64/include/generated/asm/msi.h
  WRAP    arch/arm64/include/generated/asm/preempt.h
  WRAP    arch/arm64/include/generated/asm/qrwlock.h
  WRAP    arch/arm64/include/generated/asm/rwsem.h
  WRAP    arch/arm64/include/generated/asm/segment.h
  WRAP    arch/arm64/include/generated/asm/serial.h
  WRAP    arch/arm64/include/generated/asm/set_memory.h
  WRAP    arch/arm64/include/generated/asm/sizes.h
  WRAP    arch/arm64/include/generated/asm/switch_to.h
  WRAP    arch/arm64/include/generated/asm/unaligned.h
  WRAP    arch/arm64/include/generated/asm/trace_clock.h
  WRAP    arch/arm64/include/generated/asm/user.h
  WRAP    arch/arm64/include/generated/asm/vga.h
  WRAP    arch/arm64/include/generated/asm/xor.h
  HOSTCC  scripts/kallsyms
  HOSTCC  scripts/dtc/dtc.o
  CC      scripts/mod/empty.o
  HOSTCC  scripts/mod/mk_elfconfig
  CC      scripts/mod/devicetable-offsets.s
  HOSTCC  scripts/dtc/flattree.o
  MKELF   scripts/mod/elfconfig.h
  CHK     scripts/mod/devicetable-offsets.h
  UPD     scripts/mod/devicetable-offsets.h
  HOSTCC  scripts/mod/sumversion.o
[... 134 lines cut ...]
       ^
/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/Vayu-stock/arch/arm64/kernel/stacktrace.c:246:24: error: incomplete definition of type 'struct stack_trace'
                        trace->entries[trace->nr_entries++] = frame.lr;
                                       ~~~~~^
/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/Vayu-stock/include/linux/stackdepot.h:26:8: note: forward declaration of 'struct stack_trace'
struct stack_trace;
       ^
/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/Vayu-stock/arch/arm64/kernel/stacktrace.c:[25](https://github.com/Fighteros/Android-Kernel-Builder/actions/runs/11988532933/job/33423908172#step:9:26)2:6: error: expected identifier or '('
void save_stack_trace_user(struct stack_trace *trace)
     ^
/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/Vayu-stock/include/linux/stacktrace.h:38:41: note: expanded from macro 'save_stack_trace_user'
# define save_stack_trace_user(trace)                   do { } while (0)
                                                        ^
/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/Vayu-stock/arch/arm64/kernel/stacktrace.c:252:6: error: expected identifier or '('
/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/Vayu-stock/include/linux/stacktrace.h:38:48: note: expanded from macro 'save_stack_trace_user'
# define save_stack_trace_user(trace)                   do { } while (0)
                                                               ^
10 errors generated.
make[2]: *** [/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/Vayu-stock/scripts/Makefile.build:[36](https://github.com/Fighteros/Android-Kernel-Builder/actions/runs/11988532933/job/33423908172#step:9:37)1: arch/arm64/kernel/stacktrace.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/Vayu-stock/Makefile:1189: arch/arm64/kernel] Error 2
make[1]: Leaving directory '/home/runner/work/Android-Kernel-Builder/Android-Kernel-Builder/Vayu-stock_vayu/out'
make: *** [Makefile:146: sub-make] Error 2
Error: Process completed with exit code 2.
```

winbo-yml-exe, 2024-11-23:
their kernels are incomplete lmfao

Fighteros, 2024-11-24:
> their kernels are incomplete lmfao

Oh i see that's why the kernel doesn't compile ?

Morrigan-Ship, 2025-05-18:
> their kernels are incomplete lmfao

then what should i do?
please help

komaru-meow, 2025-05-18:
> > their kernels are incomplete lmfao
> 
> then what should i do? please help

simple, email the FSF or sue xiaomi yourself if you really want to

they are required by law to give you the actual source code they use

Morrigan-Ship, 2025-05-18:
> they are required by law to give you the actual source code they use

did you traid this way?
if yes then...?

Morrigan-Ship, 2025-05-19:
> > > they are required by law to give you the actual source code they use
> > 
> > 
> > did you traid this way? if yes then...?
> 
> tried, it's why the gale branch exists..

what is gale branch?