MiCode/Xiaomi_Kernel_OpenSource #36488: Request for Kernel Source Code Release for Redmi 13C (GALE)

Andy7710, 2024-11-25:
Dear Xiaomi Support Team,

I hope this message finds you well. I am writing to request the release of the kernel source code for the Redmi 13C (GALE). As an active member of the developer community, I understand the importance of compliance with the GNU General Public License (GPL) and the positive impact it has on both users and developers.

Releasing the kernel source code for the Redmi 13C will not only help the development of custom ROMs and kernels but also foster innovation and improve the overall user experience for Xiaomi customers. It would greatly benefit the community and enhance the capabilities of the device.

I kindly request that you make the kernel source code for the Redmi 13C available to the public as soon as possible. Your cooperation in this matter is highly appreciated.

Thank you for your attention and support.

Best regards,

Andy

QAQ-cmd, 2024-11-25:
13R和13C应该用的一样的内核，，，
https://github.com/MiCode/kernel_build/tree/breeze-u-oss

QAQ-cmd, 2024-11-25:
> […](#)

可以直接用的  硬件都一样的 应该只是销售地区不同所以命名不同

Andy7710, 2024-11-25:
Pero el redmi 13r tiene un procesador Qualcomm y el redmi 13C Gale tiene un
procesador mediatek

Andy7710, 2024-11-25:
Están seguros de que ese es el kernel porque estos dos teléfonos son
diferentes en hadware