MiCode/Xiaomi_Kernel_OpenSource #38499: Release kernel source code for Emerald (Poco M6 Pro 4G/Redmi Note 13 Pro 4G), please.

DG5312, 2025-01-16:
This device have a helpful and awesome community in Telegram, the release of kernel would help us to improve and make our attempts to make a rom succesful and give the opportunity to the community for improve this aspect which several phones already have.

We love this phone, their 200 MP camera is awesome.

Please, share the kernel source code 🙂

[image]

benjaluth09, 2025-01-19:
We'd absolutely love to have the kernel for these devices! Custom ROM performance and integration would be further improved with the release of its kernel.

spixi, 2025-04-21:
The codename for Redmi Note 14s for Android 14 is there: https://github.com/MiCode/Xiaomi_Kernel_OpenSource/tree/emerald_r-u-oss
The main difference between Redmi Note 13 Pro 4G and Redmi Note 14s is that the case is completely made of plastic, which makes it lighter and gives it IP64 instead of IP54. It comes with the 200 MP camera (The cheaper Poco M5 has the 64 MP camera)

closed 2025-09-08 by DG5312