MiCode/Xiaomi_Kernel_OpenSource #39073: Publish source kernel & vendor CODENAME: emerald

FaliedDedd, 2025-02-06:
Please publish source for device Redmi Note 13 Pro 4G [emerald]

Kronos256, 2025-02-08:
> I have contacted Xiaomi customer support, they said "we are bonded to the terms and conditions of the company policies". I'm assuming that they meant MediaTek's policy. Tho if it is MediaTek stopping them from publishing the code - does that mean MediaTek is (pretty sure they're doing it knowingly) breaking the GPL?

(not same device but same problem)
They answered me after I wrote them a detailed email about GNU GPL violation, and they do not care at all about it.

here the answer from them:

" **Hello, Greetings from Xiaomi Customer Support! This is Luna.
Thanks for your support with Xiaomi. It is our honor to have such a loyal Mi user like you. And we would be grateful if you could share your country/region with us.
All program and codes are our masterpieces which Xiaomi invested not only tons of money but also time to achieve. Also, to guarantee our users' privacy safety, we must protect such information from those who may have improper purposes. Hence, we can't and won't share any of our critical information to anyone or organization. We hope you could understand that it is very important to everyone. However, if you are interested in this, we would like you to follow our official Facebook https://www.facebook.com/xiaomiglobal and website http://www.mi.com/en/ where you may find activities which appeals to you. 
We hope the above information helps you.
Best Regards
Xiaomi Global Email Service
If you have any other questions, welcome to visit again. Always believe that something wonderful is about to happen! 
Follow Twitter@XiaomiSupport to get latest product tips and service information. https://twitter.com/XiaomiSupport** "

GEOEGII555, 2025-02-09:
> (not same device but same problem) They answered me after I wrote them a detailed email about GNU GPL violation, and they do not care at all about it.
>
> here the answer from them:
>
> " **Hello, [...]

I received a similar response after I told them my region. So I told them: "I would like for you to understand that your Android derivative HyperOS is based on the Linux kernel, which is licensed under the condition that you will provide the source code if you make a derivative work."