MiCode/Xiaomi_Kernel_OpenSource #39181: kernel for the Xiaomi redmi 12c (earth) does not boot.

Oscar1640, 2025-04-01:
After building the kernel and packaging it into the boot.img, it doesn't work. It loads for a few minutes, but then reboots without starting the operating system.

same problem from [here](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/4682)

Oscar1640, 2025-04-01:
[kernel.gz](https://github.com/user-attachments/files/19544923/kernel.gz)

Awol75, 2025-05-24:
> After building the kernel and packaging it into the boot.img, it doesn't work. It loads for a few minutes, but then reboots without starting the operating system.
> 
> same problem as [here](https://github.com/MiCode/Xiaomi_Kernel_OpenSource/issues/4682)

May be your phone is manufactured recently and shipped with android 14 hyper OS instead of MIUI 13, the Redmi 12C models manufactured later have new screen model different from the original one, and the old kernel available on GitHub lacks the necessary driver for the new screen and Xiaomi unfortunately is not updating the source code. The new screen is not compatible with both stock MIUI 13 and 14, I don't think Xiaomi will update the source code as they just ignore all issues and requests