MiCode/Xiaomi_Kernel_OpenSource #39300: [要求] Redmi Note 13 4G (sapphire) GNU 内核源代码(kernel request)

Vvsxr, 2025-04-01:
亲爱的小米/Redmi团队：

Redmi Note 13 4G（代号：sapphire）发布至今已有一年零两个月，作为开源社区的一员，我们敦促贵公司在GNU通用公共许可证（GPL）下发布手机源代码。

社区中有很多人都提出了这个问题，我们希望贵公司能立即采取行动，支持开发者和爱好者通过改进软件来改进设备。

感谢您的关注和理解。

Dear Xiaomi/Redmi Team:
It has been a year and 2 months since the release of Redmi Note 13 4G (codename: sapphire). As members of the open source community, we urge your company to release the source code of the phone under the GNU General Public License
(GPL).

Many people in the community have raised this issue, and we hope that your company can take immediate action on this matter and support developers and enthusiasts to improve the device by improving the software.

Thank you for your attention and understanding.

EkinStrop, 2025-04-01:
Stop spamming.

d3formed, 2025-04-02:
bro nuked the issue page for atleast 5 pages rn.

Oscar1640, 2025-04-02:
stop spamming, they should block you.

Biraru, 2025-04-05:
Yo, dont spam. it reduces our chance of getting the kernel source.