MiCode/Xiaomi_Kernel_OpenSource #39304: We haven't got a kernel source

ximerro, 2025-04-11:
Hello, why didn't you post the kernel source for Xiaomi 11T? According to the terms of the GNU GPL license, you are required to post the kernel sources. If you do not do this, I will file a complaint against you with the Free Software Foundation.

mant1zz, 2025-04-11:
I m user mi 11t, and I want kernel for my device too

tikaaziza, 2025-04-28:
> Hello, why didn't you post the kernel source for Xiaomi 11T? According to the terms of the GNU GPLv2 license, you are required to post the kernel sources. If you do not do this, I will file a complaint against you with the Free Software Foundation.

🤣🤣🤣🤣 cmon, do u think xiaomi affraid

Stephanizgo, 2025-05-29:
I have been waiting for so long i switched my device twice and still no kernel source for 11T, what a disappointment