MiCode/Xiaomi_Kernel_OpenSource #39321: Request for Kernel Source Release for Redmi Note 13 4G (Sapphire/Sapphiren)

Angelpro09xd, 2025-05-22:
Dear Xiaomi MiCode Team,

We, as part of the development community for the Redmi Note 13 4G (Sapphire/Sapphiren), respectfully request the release of the kernel source code (Kernel OSS) for this device.

Access to the kernel source is essential for developing custom kernels and AOSP-based custom ROMs. Without it, developers are unable to compile properly functioning kernels or bring stable custom Android builds to the device, severely limiting its development potential.

The Redmi Note 13 4G has a growing user base and strong interest from the community in contributing to its software development. However, the absence of the kernel source blocks any serious work on third-party ROMs or performance enhancements.

We kindly ask you to release the complete and official kernel source code for Sapphire/Sapphiren, in line with Xiaomi’s past support for open source on other devices.

The community is ready and eager to contribute. With your support, we can unlock the true potential of this device.

Sincerely,
The custom ROM and kernel development community for the Redmi Note 13 4G
#ReleaseSapphireKernel
#XiaomiOSS

jkriyabd, 2026-10-05:
we haven’t got HO3 update, insted of this provide us the ''Kernel Source'' atleast

pbzin, 2026-10-05:
> Não recebemos a atualização HO3, em vez disso, forneça-nos pelo menos o código-fonte do kernel.

Xiaomi devs are completely ignoring our device.