MiCode/Xiaomi_Kernel_OpenSource #39329: Xiaomi 12S Ultra (thor) Kernel Source request

cornshiba, 2025-05-29:
This device has been released almost 3 years ago. Yet, there is no kernel source for the Xiaomi 12S Ultra while the kernel source for the Xiaomi 15 Ultra has already been released, and devices around this model have gotten their kernel source released including but not limited to: Xiaomi 12, Xiaomi 12 pro, Xiaomi 12S and etc.

RwOnke, 2025-06-13:
100% email them

cornshiba, 2025-06-18:
> You want me to email the FSF about this? I'll happily do that.

Hi, yes, please do that. Sorry for my late reply, but please keep me updated about this.

cornshiba, 2025-06-22:
> Sent the email, hope they don't ignore it like how they ignored me last time...

I hope so too, thank you. I just want the kernel source released

cornshiba, 2025-06-30:
> FSF didn't reply... I got a next plan. Threatening and spamming their issues. This is the most common reason to Xiaomi releasing kernel sources.

Let's do it