MiCode/Xiaomi_Kernel_OpenSource #40276: Request for Release of Kernel Source Code for Redmi Note 14 4G (tanzanite) – Compliance with GPLv2 License

muzammalvoltic, 2025-07-30:
Dear Xiaomi Development Team,

I am writing to request the immediate release of the kernel source code and device tree for the **Redmi Note 14 4G**. This request is in line with the requirements set forth by the **GPLv2 (General Public License version 2)** under which Android is distributed. As per the GPLv2, any modifications made to the Linux kernel, including those integrated into Android devices, must have the corresponding source code made available to the public.

As of now, the kernel source code for this device has not been released, which places Xiaomi in violation of the GPLv2 license. This license mandates that the source code for the kernel, which is essential for custom ROM development and proper device support, must be made available to the open-source community.

If the source code is not released promptly, it may result in legal consequences, as the GPLv2 requires adherence to its terms. The open-source community, developers, and users who rely on Xiaomi's devices for custom development and improvements are being directly impacted.

I respectfully request that Xiaomi **immediately** release the kernel source code and device tree for the **Redmi Note 14 4G** in compliance with the GPLv2 license. Failure to do so will force the community to explore other options, including legal avenues, to ensure compliance with open-source licensing standards.

I look forward to your prompt response and hope Xiaomi will take the necessary steps to support the open-source community by releasing the required source code.

Thank you for your attention to this matter.

Sincerely,
Muzammal

CharaDreemurrDT, 2025-10-17:
Hey everyone, i know it's been long since this issue has been active.
I found some info about tanzanite development, and it's quite active, just hidden in a telegram group. 
There's lot of ROMS, Custom ROMS, ports of Stock ROMS from other manufacturers.
I am using PixelOS right now and it's quite stable.
Here is the telegram group: https://t.me/RN14ID
And also there's a device tree for tanzanite right here: https://github.com/rexprjkt/android_device_xiaomi_tanzanite
Tanzanite is GKI 2.0 compliant, so it can use GKI kernels from AOSP or any other source.

muzammalvoltic, 2025-10-18:
Thanks for the info, im already in these communities, but its more of a port not the exact kernel or DT, but still good enough, i really hope they will release source code for this phone.

mian196, 2026-04-24:
> I have submitted a formal GPL source request to Xiaomi support on April 23, 2026. Awaiting response.

Where you submit it?

fly05896-rgb, 2026-10-23:
I hope the source code will be released as soon as possible, since the source code has already been released. Redmi note 15 When the 14 4G version is released The phone has been out for a year and eight months now, and I hope Xiaomi will release the source code as soon as possible.