MiCode/Xiaomi_Kernel_OpenSource #40302: Redmi Note 11 Pro 4G India (vida)  (kernel request)

GuildControl, 2025-09-06:
Dear MiCode Team,

I hope this message finds you well.

I am writing to kindly request the release of the kernel source code for the Redmi Note 11 Pro 4G India (codename: vida). According to the GPL requirements, publishing the corresponding kernel source ensures compliance and also greatly benefits both developers and end users.

Having the kernel sources available is essential for:

Security auditing and timely patching of vulnerabilities.

Enabling developers to build optimized custom ROMs and kernels that extend device lifespan and improve performance.

Supporting the developer community in contributing fixes, enhancements, and maintaining long-term software support.

The Redmi Note 11 Pro 4G India is a widely used device, and releasing the kernel sources will help ensure its users continue to receive a secure and up-to-date experience, even beyond official update cycles.

We deeply appreciate your continued effort in supporting the open-source community and kindly ask if you could prioritize publishing the kernel source for this model at your earliest convenience.

Thank you very much for your attention and support.

Best regards,
Sadly Redmi  Note 11 Pro 4G community :(

hbvvipx, 2025-09-06:
Well done

hbvvipx, 2025-09-06:
Keep moving forward

udumb4r4, 2025-09-06:
yes, where is kernel source for 4.19 version 🔥

hbvvipx, 2025-09-06:
> yeah please release source code for kernel 4.19.191

I don't know these programming things.

udumb4r4, 2025-09-06:
> > yeah please release source code for kernel 4.19.191
> 
> I don't know these programming things.

something like cooking raw material

Parviz-sudo, 2025-09-07:
Please release source code for kernel 4.19.191