MiCode/Xiaomi_Kernel_OpenSource #40326: Subject: Request for Release of Redmi 14C Kernel Source (Android 15)

Programmer-Shakib, 2025-09-19:
Dear Xiaomi Team,

I am an independent developer and enthusiast working with the Redmi 14C. In order to contribute to the custom ROM community and expand development for this device, I would like to request the official kernel source for the Redmi 14C running Android 15 (Hyperos/stock builds).

Access to the kernel source is crucial for compiling custom ROMs, optimizing performance, and ensuring compliance with the GPL licensing terms. Releasing the source will not only support developers but also encourage broader community contributions and enhance the overall software ecosystem for Redmi 14C users.

I would greatly appreciate your support in making the kernel source publicly available.

Thank you for your time and assistance.

Sincerely,
[Shakib]
Xiaomi Community

Programmer-Shakib, 2025-10-04:
There's 15C kernel source but not 14C that's unfair

j-a-sunny, 2026-10-24:
Is there any hope left?